turtlebot/turtlebot_apps #143: Cannot build turtlebot_apps package in indigo branch

nampi, 2016-03-04:
1. I created a workspace my_ws/src/
2. Cloned turtlebot_apps in src
3. In my_ws did catkin_make
I had an error. When I added "CATKIN_IGNORE" files in "software/pano" packages it worked.
Could you check, what is a problem with pano?

tfoote, 2016-03-04:
It's successfully building on the build farm. I'd need to see the error to say more, but my guess is that you are missing a dependency for it from your system. 

`rosdep install --from-path WORKSPACE --ignore-src` where you replace WORKSPACE will fix theproblem if that's the problem. see also http://answers.ros.org/question/75241/install-ros-dependencies-from-apt/

nampi, 2016-03-04:
```
In file included from /home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/src/feature_utils.cpp:8:0:
/home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/include/pano_core/feature_utils.h:101:22: error: ‘FeatureDetector’ was not declared in this scope
  std::vector<cv::Ptr<FeatureDetector> > detectors_;
                      ^
/home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/include/pano_core/feature_utils.h:101:22: note: suggested alternative:
In file included from /opt/ros/indigo/include/opencv-3.1.0-dev/opencv2/features2d/features2d.hpp:48:0,
                 from /home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/include/pano_core/feature_utils.h:12,
                 from /home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/src/feature_utils.cpp:8:
/opt/ros/indigo/include/opencv-3.1.0-dev/opencv2/features2d.hpp:206:19: note:   ‘cv::FeatureDetector’
 typedef Feature2D FeatureDetector;
                   ^
In file included from /home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/src/feature_utils.cpp:8:0:
/home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/include/pano_core/feature_utils.h:101:37: error: template argument 1 is invalid
  std::vector<cv::Ptr<FeatureDetector> > detectors_;
                                     ^
/home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/include/pano_core/feature_utils.h:101:39: error: template argument 1 is invalid
  std::vector<cv::Ptr<FeatureDetector> > detectors_;
                                       ^
/home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/include/pano_core/feature_utils.h:101:39: error: template argument 2 is invalid
/home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/include/pano_core/feature_utils.h: In constructor ‘pano::GriddedDynamicDetectorAdaptor::GriddedDynamicDetectorAdaptor(int, int, int, int, const Adjuster&)’:
/home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/include/pano_core/feature_utils.h:86:40: error: request for member ‘size’ in ‘((pano::GriddedDynamicDetectorAdaptor*)this)->pano::GriddedDynamicDetectorAdaptor::detectors_’, which is of non-class type ‘int’
   for (int i = 0; i < (int) detectors_.size(); ++i) {
                                        ^
/home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/include/pano_core/feature_utils.h:87:16: error: invalid types ‘int[int]’ for array subscript
    detectors_[i] = new cv::DynamicAdaptedFeatureDetector(new Adjuster(adjuster),maxPerCell
                ^
/home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/include/pano_core/feature_utils.h:87:24: error: expected type-specifier
    detectors_[i] = new cv::DynamicAdaptedFeatureDetector(new Adjuster(adjuster),maxPerCell
                        ^
/home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/include/pano_core/feature_utils.h:87:24: error: expected ‘;’
/home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/src/feature_utils.cpp: In member function ‘virtual void pano::GriddedDynamicDetectorAdaptor::detectImpl(const cv::Mat&, std::vector<cv::KeyPoint>&, const cv::Mat&) const’:
/home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/src/feature_utils.cpp:33:34: error: invalid types ‘const int[int]’ for array subscript
       detectors_[i * gridCols + j]->detect(sub_image, sub_keypoints, sub_mask);
                                  ^
[  2%] Built target _pano_ros_generate_messages_check_deps_PanoCaptureResult
/home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/src/Images.cpp: In member function ‘virtual void pano::Images::deserialize(const cv::FileNode&)’:
/home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/src/Images.cpp:102:13: error: ‘string’ was not declared in this scope
   fname_ = (string)fn["fname"];
             ^
/home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/src/Images.cpp:102:13: note: suggested alternative:
In file included from /usr/include/c++/4.8/string:39:0,
                 from /opt/ros/indigo/include/opencv-3.1.0-dev/opencv2/core/cvstd.hpp:58,
                 from /opt/ros/indigo/include/opencv-3.1.0-dev/opencv2/core/base.hpp:56,
                 from /opt/ros/indigo/include/opencv-3.1.0-dev/opencv2/core.hpp:54,
                 from /opt/ros/indigo/include/opencv-3.1.0-dev/opencv2/core/core.hpp:48,
                 from /home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/include/pano_core/Images.h:11,
                 from /home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/src/Images.cpp:1:
/usr/include/c++/4.8/bits/stringfwd.h:62:33: note:   ‘std::string’
   typedef basic_string<char>    string;   
                                 ^
/home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/src/Images.cpp:102:20: error: expected ‘;’ before ‘fn’
   fname_ = (string)fn["fname"];
                    ^
/home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/src/Images.cpp:103:19: error: expected ‘;’ before ‘fn’
   path_ = (string)fn["path"];
                   ^
/home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/src/Images.cpp: In member function ‘cv::Mat pano::HugeImage::loadAll() const’:
/home/nadya/my_ws/src/turtlebot_apps/software/pano/pano_core/src/Images.cpp:114:12: error: ‘string’ was not declared in this scope
   map<int, string>::const_iterator nit = names_.begin();
            ^
[... cut ...]
[  2%] Built target laser_footprint_filter
make[2]: *** [turtlebot_apps/software/pano/pano_core/CMakeFiles/pano_core.dir/src/ModelFitter.cpp.o] Error 1
[  2%] [  3%] make[1]: *** [turtlebot_apps/software/pano/pano_core/CMakeFiles/pano_core.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Built target turtle_stroll
Built target _turtlebot_calibration_generate_messages_check_deps_ScanAngle
[ 12%] Built target pano_ros_generate_messages_py
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

```

stonier, 2016-03-08:
Some of those `FeatureDetector` references really ought to be prefixed with a `cv::` namespace. Perhaps you are using a compiler which is stricter in the lookups than whatever is building for us on the build farm for some reason?

What OS/g++ are you using?

130s, 2016-06-16:
Just FYI. Although I'm not sure I'm on the same platform as OP was (the log says /opt/ros/indigo but which OS?), I was seeing the similar/same error issue on Kinetic-Xenial, and with `CATKIN_IGNORE` I'm now able to build with the following tweaks:

- CATKIN_IGNORE-d packages:
 - `pano_*`
 - `turtlebot_panorama` (due to pano_* not building)
 - `turtlebot_rapps` (due to `warrehouse_ros` not building on Kinetic. FYI if interested https://github.com/ros-planning/warehouse_ros/pull/32#issuecomment-226566531)
- Use `catkin_make_isolated`. `catkin build` seems to capture more errors as @stonier pointed out.
- I used this source tree. Everything else is relied on DEBs:

  ```
$ tree -L 2 src
src
├── CMakeLists.txt -> /opt/ros/kinetic/share/catkin/cmake/toplevel.cmake
├── corot
│   ├── world_canvas
│   ├── world_canvas_libs
│   └── world_canvas_msgs
├── GT-RAIL
│   └── robot_pose_publisher
├── robotics-in-concert
│   ├── rocon_app_platform
│   └── rocon_multimaster
├── turtlebot
│   ├── turtlebot
│   ├── turtlebot_apps
│   └── turtlebot_simulator
├── yujinrobot
│   ├── kobuki
│   ├── kobuki_desktop
│   └── yujin_ocs

```

rohbotics, 2016-12-19:
As the panorama stuff was removed, and spun out into a separate repo, I think this can be closed now.

closed 2016-12-20 by tfoote