turtlebot/turtlebot_apps #159: No such file or directory amcl.launch.xml under turtlebot_navigation package!

mzaiady, 2016-12-07:
I'm getting the following error

    
> while processing /opt/ros/indigo/share/turtlebot_navigation/launch/includes/amcl.launch.xml: Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/opt/ros/indigo/share/turtlebot_navigation/launch/includes/amcl.launch.xml' The traceback for the exception was written to the log file

While running command

`roslaunch turtlebot_gazebo amcl_demo.launch`

mzaiady, 2016-12-07:
I used the following workaround until turtlebot_gazebo,  by  updating the 
  

`/opt/ros/indigo/share/turtlebot_gazebo/launch/amcl_demo.launch`

To the following (same logic included in the updated  turtlebot_navigation/launch/amcl_demo.launch ), 

 

        ```
<launch>
      
      <arg name="map_file" default="$(env TURTLEBOT_GAZEBO_MAP_FILE)"/>
      <arg name="3d_sensor" default="$(env TURTLEBOT_3D_SENSOR)"/>  
    
      <node name="map_server" pkg="map_server" type="map_server" args="$(arg map_file)" />
    
      
      <arg name="initial_pose_x" default="0.0"/>
      <arg name="initial_pose_y" default="0.0"/>
      <arg name="initial_pose_a" default="0.0"/>
     <arg name="custom_amcl_launch_file" default="$(find turtlebot_navigation)/launch/includes/amcl/$(arg 3d_sensor)_amcl.launch.xml"/> 
    
      <include file="$(arg custom_amcl_launch_file)">
        <arg name="initial_pose_x" value="$(arg initial_pose_x)"/>
        <arg name="initial_pose_y" value="$(arg initial_pose_y)"/>
        <arg name="initial_pose_a" value="$(arg initial_pose_a)"/>
      </include>
    
      
     <arg name="custom_param_file" default="$(find turtlebot_navigation)/param/$(arg 3d_sensor)_costmap_params.yaml"/>*
     <include file="$(find turtlebot_navigation)/launch/includes/move_base.launch.xml">
     <arg name="custom_param_file" value="$(arg custom_param_file)"/>
     </include>
    </launch>

```
Replaced 

   ` <include file="$(find turtlebot_navigation)/launch/includes/amcl.launch.xml">`

With:

  ```
  <arg name="3d_sensor" default="$(env TURTLEBOT_3D_SENSOR)"/> 
    <arg name="custom_amcl_launch_file" default="$(find turtlebot_navigation)/launch/includes/amcl/$(arg 3d_sensor)_amcl.launch.xml"/> 
      <include file="$(arg custom_amcl_launch_file)">
```

also added 
`<arg name="custom_param_file" default="$(find turtlebot_navigation)/param/$(arg 3d_sensor)_costmap_params.yaml"/>*`
`<arg name="custom_param_file" value="$(arg custom_param_file)"/>`

mzaiady, 2016-12-07:
The issue also exist if you creating a new map, so to make my answer complete, also gmapping-demo.launch need to be updated:

` sudo gedit /opt/ros/indigo/share/turtlebot_gazebo/launch/gmapping_demo.launch`

and replace the content with :

```
<launch>
  <arg name="3d_sensor" default="$(env TURTLEBOT_3D_SENSOR)"/>  <!-- r200, kinect, asus_xtion_pro -->
  <arg name="custom_gmapping_launch_file" default="$(find turtlebot_navigation)/launch/includes/gmapping/$(arg 3d_sensor)_gmapping.launch.xml"/>
  <include file="$(arg custom_gmapping_launch_file)"/>
</launch>
```

kevincwells, 2016-12-07:
Good catch -- yes, the turtlebot_gazebo launch files should be updated to point to the right location.  Apologies for this oversight.  If you've made the changes, can you submit them as a pull request?

mzaiady, 2016-12-07:
I submitted one, 10 minutes ago.

rohbotics, 2016-12-19:
This can be closed, @mzaiady 's PR was merged.

https://github.com/turtlebot/turtlebot_simulator/pull/61

stonier, 2016-12-19:
Thanks for tracking some of these @rohbotics

closed 2016-12-19 by stonier

kevincwells, 2016-12-20:
Yeah, @rohbotics has been on a roll today. Thanks for testing so many of these issues and pull requests!

rohbotics, 2016-12-20:
Thanks! :blush: 
I am interning at OSRF for my holiday break this week, so I started by going through pending PRs for turtlebot things.