turtlebot/turtlebot_apps #173: Integrated ZED stereo camera

willzoe, 2019-12-12:
CUDA: 9.0
ZED SDK version: 2.8.3
Ubuntu 16.04
ROS Kinetic

I have a ZED stereo camera.
How to replace 3D sensors such as kinect, integrate ZED into the navigation package, and implement turtlebot2 mapping and navigation？

yuanzhedong, 2020-08-27:
Hi @willzoe, do you find the solution? I'm also trying the ZED camera.

willzoe, 2020-08-27:
Yes. I have integrated ZED camera on turtlebot2, and implemented mapping and location using RTAB-Map.

yuanzhedong, 2020-08-31:
> Yes. I have integrated ZED camera on turtlebot2, and implemented mapping and location using RTAB-Map.

That's great, is there any documentation you could share? Really appreciate it!

willzoe, 2020-09-01:
I have not sorted out the relevant documents. If you encounter some problems, I can do my best to answer you.

yuanzhedong, 2020-09-01:
Got it, thanks!